khoih-prog/MDNS_Generic #1: Support IndustrialShields mDuino family of Confrollers

tejpochiraju, 2021-07-22:
**Is your feature request related to a problem? Please describe.**
I am trying to use this library with a IndustrialShields mDuino-21+ controller (based on a Mega 2560) and keep running into a compilation error when building the `DiscoveringServices.ino` example. 

```
/home/tej/Arduino/libraries/MDNS_Generic/src/MDNS_Generic_Impl.h:177:30: error: 'class UDP' has no member named 'beginMulticast'; did you mean 'beginPacket'?
     statusCode = this->_udp->beginMulticast(mdnsMulticastIPAddr, MDNS_SERVER_PORT);
...
Using library SPI at version 1.0 in folder: /home/tej/.arduino15/packages/industrialshields/hardware/avr/1.1.35/libraries/SPI 
Using library Ethernet at version 2.0.0 in folder: /home/tej/.arduino15/packages/industrialshields/hardware/avr/1.1.35/libraries/Ethernet 
Using library MDNS_Generic at version 1.1.0 in folder: /home/tej/Arduino/libraries/MDNS_Generic 
Compilation error: Error: 2 UNKNOWN: exit status 1
```

The same code builds and runs fine when I select `Mega` as the board within the IDE. However, I need to merge this into an existing project that depends on the mDuino platform files. 

**Describe the solution you'd like**
Support the mDuino family of controllers OR share instructions on how to port to new controllers. 

**Describe alternatives you've considered**
- I tried to read through the LibraryPatches to understand the changes being made.
- I tried replacing the files in `Ethernet/src` to see if that would help.

**Additional context**
The IndustrialShields Platform can be added using [this JSON](http://apps.industrialshields.com/main/arduino/boards/package_industrialshields_index.json).

khoih-prog, 2021-07-22:
Thanks for your interest in the library.

The old and memory-constraint Mega2560 boards are not supported and I'll won't spend time to add them to the currently long list. It's better that you use one of the better and currently supported, powerful yet cheaper boards, such as ESP32, WT32_ETH01, RP2040, etc.

Good Luck,

closed 2021-07-22 by khoih-prog

tejpochiraju, 2021-07-23:
@khoih-prog - I understand and we use ESP32 in most of our work. However, in this case, we need to work with an existing controller. Since the Mega works fine, I would have expected the controller based on it to work too. 

However, I believe I am missing something in the patches. If you could share instructions on how you generate the patches I can probably make the fixes myself.

khoih-prog, 2021-07-24:
Good news is that the library is currently supporting AVR Mega 2560. There are too many boards and I can't have a good list of all.

[image]

I think you forget to apply the necessary patch for the Ehernetx library.
Follow correctly the instruction in [**Libraries' Patches**](https://github.com/khoih-prog/MDNS_Generic#libraries-patches)

khoih-prog, 2021-07-24:
The **IndustrialShields mDuino** is not supported as there are too many different boards / cores in the world designed not compatible with the Arduino AVR core.

If you're using one of those incompatible cores, you have to modify the core or read and understand the core to modify the open-sourced library to add the support.

I'm sorry I can't spend time to do that for you. You can ask the manufacturer and/or the forum for help.

khoih-prog, 2021-07-24:
OK, you can fix the **IndustrialShields mDuino** core as follows

- Replacing the file `Udh.h` (**~/.arduino15/packages/industrialshields/hardware/avr/1.1.36/cores/industrialshields/Udp.h**) with the following

```
/*
 *  Udp.cpp: Library to send/receive UDP packets.
 *
 * NOTE: UDP is fast, but has some important limitations (thanks to Warren Gray for mentioning these)
 * 1) UDP does not guarantee the order in which assembled UDP packets are received. This
 * might not happen often in practice, but in larger network topologies, a UDP
 * packet can be received out of sequence. 
 * 2) UDP does not guard against lost packets - so packets *can* disappear without the sender being
 * aware of it. Again, this may not be a concern in practice on small local networks.
 * For more information, see http://www.cafeaulait.org/course/week12/35.html
 *
 * MIT License:
 * Copyright (c) 2008 Bjoern Hartmann
 * Permission is hereby granted, free of charge, to any person obtaining a copy
 * of this software and associated documentation files (the "Software"), to deal
 * in the Software without restriction, including without limitation the rights
 * to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
 * copies of the Software, and to permit persons to whom the Software is
 * furnished to do so, subject to the following conditions:
 * 
 * The above copyright notice and this permission notice shall be included in
 * all copies or substantial portions of the Software.
 * 
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 * FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
 * AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 * OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
 * THE SOFTWARE.
 *
 * bjoern@cs.stanford.edu 12/30/2008
 */

#ifndef udp_h
#define udp_h

#include <Stream.h>
#include <IPAddress.h>

class UDP : public Stream {

public:
  virtual uint8_t begin(uint16_t) =0;	// initialize, start listening on specified port. Returns 1 if successful, 0 if there are no sockets available to use
  virtual uint8_t beginMulticast(IPAddress, uint16_t) { return 0; }  // initialize, start listening on specified multicast IP address and port. Returns 1 if successful, 0 on failure
  virtual void stop() =0;  // Finish with the UDP socket

  // Sending UDP packets
  
  // Start building up a packet to send to the remote host specific in ip and port
  // Returns 1 if successful, 0 if there was a problem with the supplied IP address or port
  virtual int beginPacket(IPAddress ip, uint16_t port) =0;
  // Start building up a packet to send to the remote host specific in host and port
  // Returns 1 if successful, 0 if there was a problem resolving the hostname or port
  virtual int beginPacket(const char *host, uint16_t port) =0;
  // Finish off this packet and send it
  // Returns 1 if the packet was sent successfully, 0 if there was an error
  virtual int endPacket() =0;
  // Write a single byte into the packet
  virtual size_t write(uint8_t) =0;
  // Write size bytes from buffer into the packet
  virtual size_t write(const uint8_t *buffer, size_t size) =0;

  // Start processing the next available incoming packet
  // Returns the size of the packet in bytes, or 0 if no packets are available
  virtual int parsePacket() =0;
  // Number of bytes remaining in the current packet
  virtual int available() =0;
  // Read a single byte from the current packet
  virtual int read() =0;
  // Read up to len bytes from the current packet and place them into buffer
  // Returns the number of bytes read, or 0 if none are available
  virtual int read(unsigned char* buffer, size_t len) =0;
  // Read up to len characters from the current packet and place them into buffer
  // Returns the number of characters read, or 0 if none are available
  virtual int read(char* buffer, size_t len) =0;
  // Return the next byte from the current packet without moving on to the next byte
  virtual int peek() =0;
  virtual void flush() =0;	// Finish reading the current packet

  // Return the IP address of the host who sent the current incoming packet
  virtual IPAddress remoteIP() =0;
  // Return the port of the host who sent the current incoming packet
  virtual uint16_t remotePort() =0;
protected:
  uint8_t* rawIPAddress(IPAddress& addr) { return addr.raw_address(); };
};

#endif
```

I'll add the patch into [**Packages_Patches**](https://github.com/khoih-prog/MDNS_Generic/tree/master/Packages_Patches) in next release.

khoih-prog, 2021-07-24:
Compile is OK with MDuino 21+ now

[image]

tejpochiraju, 2021-07-27:
Thanks, @khoih-prog - can confirm this works fine.