khoih-prog/MDNS_Generic #4: Portenta H7 MDNS Library compatibility

Zuroy, 2021-08-31:
### Discussed in https://github.com/khoih-prog/MDNS_Generic/discussions/3

<div type='discussions-op-text'>

<sup>Originally posted by **Zuroy** August 31, 2021</sup>
Good morning everyone,
I wonder if this library is compatible with Portenta H7 board. I tryed to use it with that board to add (register) a new service on my net but when I lunch the discovery routine using python zeroconf it returns this error: 
[image]

Can you help me?
Thank you very much,

Amedeo</div>

closed 2021-08-31 by khoih-prog

Zuroy, 2021-09-06:
The problem is not the Zeroconf implementation in Python. 
Using Wireshark the packets coming from the board in which is implemented the MDNS library are malformed and cannot be interpreted.

khoih-prog, 2021-09-06:
This library is currently not supporting Portenta H7. I'm surprised that it can run or do anything for you.

Zuroy, 2021-09-06:
Thank you for the reply. I'm trying to solve the compatibility problems.

khoih-prog, 2021-09-06:
> As I have no Portenta H7 board now to test, I don't know when I can add the support in the future.

Certainly if you're patience and willing to help in testing, I'll port the library to Portenta H7, then you do all the testing, bug reporting and final approving.

Zuroy, 2021-09-06:
Sure! Thank you very much.

khoih-prog, 2021-09-06:
Great.  But out of my curiosity, I already ordered a Portenta H7 to test and write / port many of my libraries to support it.

You will see soon a topic similar to this [**Libraries updated to support new RP2040-based boards (RASPBERRY_PI_PICO, etc.)**](https://forum.arduino.cc/t/libraries-updated-to-support-new-rp2040-based-boards-raspberry-pi-pico-etc/860353)

khoih-prog, 2021-09-08:
I'm sorry to let you know somehow the `mbed_portenta` core doesn't fully support **UDP multicast**, because of some bugs, designs, etc. deep down in the core. I already wasted nearly several hrs trying to isolate the issue, but still haven't found out the culprit and/or the fix.

I have to move on to continue with other more important and interesting project for **Portenta_H7** and don't know when I will spend time again with this MDNS **Portenta_H7** support.

FYI,  possibly you already knew that I started the new topic [**Libraries updated to support new Portenta H7 boards**](https://forum.arduino.cc/t/libraries-updated-to-support-new-portenta-h7-boards/903361). Follow up there to have an update of which of my libraries supporting this new, beautiful  and powerful **Portenta_H7**

Zuroy, 2021-09-09:
I appriciate very much your efforts to solve these bugs. Thank you. 
I'm following your new post and let me know if you need some help for testing or other.

Zuroy, 2021-09-09:
> I'm sorry to let you know somehow the `mbed_portenta` core doesn't fully support **UDP multicast**, because of some bugs, designs, etc. deep down in the core. I already wasted nearly several hrs [...]

Arduino support team confirms that **UDP multicast is not supported** while plain UDP should work correctly.

khoih-prog, 2021-09-09:
> Arduino support team confirms that UDP multicast is not supported while plain UDP should work correctly.

This is what I tested and found, then tried to fix, but without success yet. Thanks for the info.

I'm now working to port all of my libs to Portenta (for fun), and currently have not prioritize which one, possibly **WebSockets, TimerInterrupt, FlashStorage/LittleFS, DRD/MRD, WiFi/Ethernet Manager, MySQL, MQTT-Broker, etc.**
If you think which one is urgent for you, I can help prioritize it.

Certainly if you can **test and find bugs** in the currently supporting list, I'd appreciate it.

khoih-prog, 2021-09-09:
Hi @Zuroy 

As Portenta is not well supported yet, it's time-consuming to find libraries, examples, sample codes for many basic features.

I many times have to spend time to search around, and look deep into the core to learn and figure out.

If you're willing to help here, in **researching, locating, posting your experience**, I'd really appreciate. I can invite you into my private tech discussion repo., so that we can share / post lengthy topics unrelated to this MDNS library. Just let me know.

Zuroy, 2021-09-13:
> Hi @Zuroy
>
> As Portenta is not well supported yet, it's time-consuming to find libraries, examples, sample codes for many basic features.
>
> I many times have to spend time to search around, and look [...]

Hi khoih-prog,
currently I'm trying to solve the problem with the UDP multiicast and the MDNS since it is mandatory for the project I'm working on. I saw this discussion https://github.com/arduino/ArduinoCore-mbed/pull/14 maybe it can help us to identify the problem (even if the issue in the previous link was releated to WiFi multicast). 
I'll be very happy to participate to your tech discussion repo.

Zuroy, 2021-09-13:
> This is what I tested and found, then tried to fix, but without success yet. Thanks for the info.
>
> I'm now working to port all of my libs to Portenta (for fun), and currently have not prioritize [...]

I forgot to say that I'm working with ethernet connection

khoih-prog, 2021-09-13:
@Zuroy

Currently UDP Multicast for Portenta_H7 (both Ethernet and WiFi) is not working.

Check

1. [**Portenta H7 UDP Multicast not working - mbed v2.4.1** #321](arduino/ArduinoCore-mbed#321)
2. [**Portenta H7 MDNS Library compatibility** #4](khoih-prog/MDNS_Generic#4)

I'll invite you to a  private repo., which has been ported to run with Portenta_H7, WiFi or Ethernet. But even with the latest patches, proposed in `ArduinoCore-mbed` [**WiFi UDP & Multicast Functionality** #14](https://github.com/arduino/ArduinoCore-mbed/pull/14), the MDNS is not working with either Ethernet or WiFi.

Testing the same library on any other supported boards (ESP32/ESP8266, nRF52, etc.) is OK.

**Already try with `ArduinoCore-mbed` mbed_portenta core from v2.0.0 up to v2.4.1 without success, with or without the patches in [**Portenta_Patches**](https://github.com/khoih-prog/General_Discussion/tree/main/MDNS_Generic_Private/Portenta_Patches).**

Any more testing or help is appreciated.

khoih-prog, 2021-09-13:
Hi @Zuroy 

Just published [**MDNS_Generic releases v1.3.0-beta1**](https://github.com/khoih-prog/MDNS_Generic/releases/tag/v1.3.0-b1) for you and other people to test, for fixing Portanta_H7  UDP Multicast bug.

You can compile and run OK with Portenta_H7, Ethernet or WiFi. The examples will run `normally !!`, just can't do anything relating to **mDNS**.

---

### Releases v1.3.0-beta1

1. Add support to Portenta_H7, using WiFi or Ethernet. **Still not working**, just for preliminary tests. Check 

* 1.1 [**Portenta H7 UDP Multicast not working - mbed v2.4.1** #321](https://github.com/arduino/ArduinoCore-mbed/issues/321)
* 1.2 [**Portenta H7 MDNS Library compatibility** #4](https://github.com/khoih-prog/MDNS_Generic/issues/4)

for latest situation.

khoih-prog, 2021-09-28:
Hi @Zuroy 

The issue has been fixed by [**SocketWrapper: UDP: send packet on endPacket(), not on write()** #337](https://github.com/arduino/ArduinoCore-mbed/pull/337), thanks to your issue opened in [**Portenta H7 UDP Multicast not working - mbed v2.4.1** #321](https://github.com/arduino/ArduinoCore-mbed/issues/321)

I'll publish a new release to add support to Portenta_H7, to replace the beta release [MDNS_Generic releases v1.3.0-b1](https://github.com/khoih-prog/MDNS_Generic/releases/tag/v1.3.0-b1)

---

Test Results

1. Using Portenta_H7 WiFi

1.1 Terminal output

```
Start WiFiRegisteringServices on PORTENTA_H7_M7
MDNS_Generic v1.3.0-b1
Attempting to connect to SSID: HueNet1
SSID: HueNet1
Local IP address: 192.168.2.94
Signal strength (RSSI):-35 dBm
Registering mDNS hostname: portenta-h7-m7
To access, using portenta-h7-m7.local
[MDNS] ::begin: UDP beginMulticast statusCode= OK
[MDNS] ::_sendMDNSMessage: xid= 0
[MDNS] ::_sendMDNSMessage: queryCount= 0
[MDNS] ::_sendMDNSMessage: answerCount= 1024
[MDNS] ::_sendMDNSMessage: authorityCount= 0
[MDNS] ::_sendMDNSMessage: additionalCount= 256
[MDNS] ::_sendMDNSMessage: statusCode = 1
```
1.2 Ping using mDNS hostname => OK

```
kh@kh-Inspiron-3593:~$ ping portenta-h7-m7.local
PING portenta-h7-m7.local (192.168.2.94) 56(84) bytes of data.
64 bytes from 192.168.2.94 (192.168.2.94): icmp_seq=1 ttl=255 time=49.4 ms
64 bytes from 192.168.2.94 (192.168.2.94): icmp_seq=2 ttl=255 time=153 ms
64 bytes from 192.168.2.94 (192.168.2.94): icmp_seq=3 ttl=255 time=73.4 ms
64 bytes from 192.168.2.94 (192.168.2.94): icmp_seq=4 ttl=255 time=95.7 ms
64 bytes from 192.168.2.94 (192.168.2.94): icmp_seq=5 ttl=255 time=119 ms
64 bytes from 192.168.2.94 (192.168.2.94): icmp_seq=6 ttl=255 time=22.0 ms
64 bytes from 192.168.2.94 (192.168.2.94): icmp_seq=7 ttl=255 time=30.4 ms
64 bytes from 192.168.2.94 (192.168.2.94): icmp_seq=8 ttl=255 time=47.7 m
```

2. Using Portenta_H7 Ethernet

2.1 Terminal output

```
Start RegisteringServices on PORTENTA_H7_M7 using Ethernet using Portenta_Ethernet Library
MDNS_Generic v1.3.0-b1
=========================
Default SPI pinout:
MOSI:8
MISO:10
SCK:9
SS:7
=========================
[MDNS] Board : PORTENTA_H7_M7 , setCsPin: 10
=========================
Currently Used SPI pinout:
MOSI:8
MISO:10
SCK:9
SS:7
=========================
Using mac index = 11
Connected! IP address: 192.168.2.87
Registering mDNS hostname: portenta-h7-m7-15
To access, using portenta-h7-m7-11.local
[MDNS] ::begin: UDP beginMulticast statusCode= OK
AddService : PORTENTA_H7_M7_mDNS_Webserver._http
[MDNS] ::_sendMDNSMessage: xid= 0
[MDNS] ::_sendMDNSMessage: queryCount= 0
[MDNS] ::_sendMDNSMessage: answerCount= 1024
[MDNS] ::_sendMDNSMessage: authorityCount= 0
[MDNS] ::_sendMDNSMessage: additionalCount= 256
[MDNS] ::_sendMDNSMessage: statusCode = 1

```

2.2 Ping using mDNS hostname => OK

```
kh@kh-Inspiron-3593:~$ ping portenta-h7-m7-15.local
PING portenta-h7-m7-15.local (192.168.2.87) 56(84) bytes of data.
64 bytes from 192.168.2.87 (192.168.2.87): icmp_seq=1 ttl=255 time=0.964 ms
64 bytes from 192.168.2.87 (192.168.2.87): icmp_seq=2 ttl=255 time=0.216 ms
64 bytes from 192.168.2.87 (192.168.2.87): icmp_seq=3 ttl=255 time=1.07 ms
64 bytes from 192.168.2.87 (192.168.2.87): icmp_seq=4 ttl=255 time=4.40 ms
64 bytes from 192.168.2.87 (192.168.2.87): icmp_seq=5 ttl=255 time=0.314 ms
64 bytes from 192.168.2.87 (192.168.2.87): icmp_seq=6 ttl=255 time=0.635 ms
64 bytes from 192.168.2.87 (192.168.2.87): icmp_seq=7 ttl=255 time=0.996 ms
64 bytes from 192.168.2.87 (192.168.2.87): icmp_seq=8 ttl=255 time=4.47 ms
```

khoih-prog, 2021-09-28:
Hi @Zuroy 

The new [**MDNS_Generic** releases v1.3.0](https://github.com/khoih-prog/MDNS_Generic/releases/tag/v1.3.0) has just been published.

Your contribution is noted in [Contributions and Thanks](https://github.com/khoih-prog/MDNS_Generic#contributions-and-thanks)

Please have full test to verify and report if there is any issue. Be sure to use the [**Packages' Patches** for mbed_portenta](https://github.com/khoih-prog/MDNS_Generic/tree/master/Packages_Patches/arduino/hardware/mbed_portenta)

Best Regards,

---

### Releases v1.3.0

1. Add support to Portenta_H7, using WiFi or Ethernet. **Already working**.

Please use the [Packages' Patches for mbed_portenta](https://github.com/khoih-prog/MDNS_Generic/tree/master/Packages_Patches/arduino/hardware/mbed_portenta)

Zuroy, 2021-09-29:
> Hi @Zuroy
>
> The new [**MDNS_Generic** releases v1.3.0](https://github.com/khoih-prog/MDNS_Generic/releases/tag/v1.3.0) has just been published.
>
> Your contribution is noted in [Contributions and [...]

Thank you very much! Great work! 
I'm testing the library and when using:

`  mdns.addServiceRecord("Arduino mDNS Webserver Example._http",
                        80,
                        MDNSServiceTCP, MyText);`

Wireshark returnes an error of malformed packet, 
[image]

The text content seems to be in the wrong place.

khoih-prog, 2021-09-29:
You're very welcome, this is the **great work of Arduino team**, not mine ;-)

Did you try to add, as @facchinm suggests, added to the sketch after Ethernet.begin()

```
  STM32_EMAC &emac = STM32_EMAC::get_instance();

  ETH_MACFilterConfigTypeDef pFilterConfig;
  HAL_ETH_GetMACFilterConfig(&emac.EthHandle, &pFilterConfig);
  pFilterConfig.PassAllMulticast = ENABLE;
  HAL_ETH_SetMACFilterConfig(&emac.EthHandle, &pFilterConfig);
```

Can you `addServiceRecord` successfully ? Is the malformed packet **random** or it happens to **every packet** ? Did you test to see the same issue happens to **WiFi** as well ?

Could you also test using another supported board to see if this a bug of this library of just **Portenta** bug?

If it's happening only to Portenta, could you **post another issue on the core** ?

Thanks,

Zuroy, 2021-09-29:
> You're very welcome, this is the **great work of Arduino team**, not mine ;-)
>
> Did you try to add, as @facchinm suggests, added to the sketch after Ethernet.begin()
>
> ```
>   STM32_EMAC &emac = [...]

I solved the problem. The first character of the textcontent represents the length of the data. Inside EthernetBonjour this first character is converted to ASCII code, so a possible solution is:
` 
int len;
  String test_str = "=v+b";
  len = test_str.length();
  char int2char_len = len;
  
  String text = int2char_len + test_str ;
  int str_len = text.length()+1;
  char char_array[str_len];
 
  text.toCharArray(char_array, str_len);
  //Serial.println(char_array);
 /* 
  MyTxt lunghezza;
  lunghezza.len = strlen("ciao");
  MyTxt stringa;
  stringa.mychar = "ciao";
  */
  server.begin();

  // Initialize the mDNS library. You can now reach or ping this
  // Arduino via the host name "arduino.local", provided that your operating
  // system is mDNS/Bonjour-enabled (such as macOS).
  // Always call this before any other method!
  mdns.begin(Ethernet.localIP(), "arduino");

  // Now let's register the service we're offering (a web service) via mDNS!
  // To do so, we call the addServiceRecord() method. The first argument is the
  // name of our service instance and its type, separated by a dot. In this
  // case, the service type is _http. There are many other service types, use
  // Google to look up some common ones, but you can also invent your own
  // service type, like _mycoolservice - As long as your clients know what to
  // look for, you're good to go.
  // The second argument is the port on which the service is running. This is
  // port 80 here, the standard HTTP port.
  // The last argument is the protocol type of the service, either TCP or UDP.
  // Of course, our service is a TCP service.
  // With the service registered, it will show up in a Bonjour-enabled web
  // browser. As an example, if you are using Apple's Safari, you will now see
  // the service under Bookmarks -> Bonjour (Provided that you have enabled
  // Bonjour in the "Bookmarks" preferences in Safari).

  
  mdns.addServiceRecord("Arduino mDNS Webserver Example._http",
                        80,
                        MDNSServiceTCP, char_array);
`
Here the analysis with wireshark:
[image]