khoih-prog/MDNS_Generic #5: mDNS stops working after two minutes

stefat, 2021-09-01:
### Information on Hw/Sw pattern
Arduino core MKR WiFi 1010 
WIFININA ver. 1.4.7
MDNS_Generic ver 1.1.0
Arduino ver 1.8.15

### Describe the bug
The hw runs a web server with mDns initialised with  hostname 'mkr_1010'.
The system does well with mDNS after reset. Using Chrome/Safari browsers on different client devices (Mac, iPhone, iPad)it quickly answers the mDNS service at  http://mkr_1010.local.
ISSUE: if left for few minutes (~more than 1-2) with no web requests the system no longer answers the http://mkr_1010.local url. The fw is running as the browser correctly opens the http://<IP address> request but no any longer the http://mkr_1010.local url.

### Steps to Reproduce
It is a steady behaviour and it is always enough to wait few minutes after the system reset to reproduce. 
### Debug
Tried to remove and re-add the mDNS service after a couple of minutes of no client requests (removeAllServiceRecords();addServiceRecord(...);): no change in behaviour.

For your convenience I also add the following Wireshark screenshot:
[image]
This screenshot shows the sniffering of the filtered ip.addr==192.168.1.138 (the system under test) coming from an hour of no web requests after the last successful http://mkr_1010.local request sent from the Chrome client. 
Before time 08:19:41.091806 (blue selected row) I sent a couple of Chrome requests with no answer after the no-requests hour delay.
At time 08:19:41.091806 I pressed the reset button on the MKR WiFi 1010 pcb and sent a new http://mkr_1010.local request from the browser.
And everything restarts working: see please from time 08:22:14 and later.
## Additional context
Googling around I've found this same issue exists for other platform(s). See please https://github.com/espressif/arduino-esp32/issues/4406

I'm desperate.
Any help very appreciated.

Thank you.

Stefat

khoih-prog, 2021-09-01:
It's seems the issue is wide-spread, and possibly caused by 

1. `avahi` bug and/or 
2.  your router (not caching MDNS data)

but hopefully, we can possibly have a work-around as follows:

Have a look at [**mDNS sends reply to multicast instead of unicast if request source port is 5353 (breaks avahi-resolve) (IDFGH-5375)** #7124](https://github.com/espressif/esp-idf/issues/7124)

> The actual difference between the dig query and the one sent by avahi is not only the source port, but mainly the fact that dig (and some other libraries, lwip, too) sends so called One shot queries defined in https://datatracker.ietf.org/doc/html/rfc6762#section-5.1, whereas avahi is a fully compliant mDNS querier.
> 
> The mDNS library doesn't (correctly) support both scenarios, trying to comply with full featured mDNS responder and at the same time responds to lwip mDNS queries introduced MDNS_REPEAT_QUERY_IN_RESPONSE and CONFIG_MDNS_STRICT_MODE configuration options.
> Enabling the strict mode in the project configuration menu would make it respond to avahi's queries (after those 2 minute TTLs), but stop responding to dig's (note, that avahi deviates from the spec here, a little, as well: https://datatracker.ietf.org/doc/html/rfc6762#page-14)
> 
> Here's a quick fix, which should work for both queriers:
> mdns-Support-for-One-Shot-mDNS-queries.patch.txt

I'll spend time to research, understand and test the fix, then implement for this library if OK.

Anything (more research, tests with different router, info, links, etc.) 

Be patient,

khoih-prog, 2021-09-01:
More info about `avahi` bug

[**avahi-daemon rejects response packet if it contains the original question record** #348](https://github.com/lathiat/avahi/issues/348)

stefat, 2021-09-01:
happy to know it's your concern as well ! I'll stay patient...and keen.
Thanks

khoih-prog, 2021-09-01:
FYI, The good explanation of the behaviour is in [ **Open mDNS stops working after two minutes** #4406](https://github.com/espressif/arduino-esp32/issues/4406)

> I can confirm that with the mDNS server in the Espressif native framework (ESP-IDF v4.2), the same pattern of packets is sent.
> At startup, after obtaining an IP address the ESP32 first broadcasts an mDNS query for the hostname it is about to use (with a **TTL of 2 minutes**). It does this three times.
> 
> If no other machine responds (which would indicate this hostname is already assigned to another machine), the ESP32 then broadcasts an mDNS response to its own query, advertising its IP address (**with a TTL of 2 mins**). It does this three times.
> 
> After this initial announcement I did not see the ESP32 send any further mDNS packets during the 30min period of my test - although it did continue to send periodic ARP broadcasts (which are not related to mDNS).
> 
> However unlike the OP, I do not lose the ability to resolve the hostname after the 2 minute period... indeed all the machines on my LAN were still able to resolve and ping the host four hours later.
> 
> Looking into this a little more closely, I can see that it is in fact my residential gateway (FritzBox 7530) that has **cached the mDNS result** for the ESP32, and is serving it to other local clients via the 'normal' DNS service on port 53.

stefat, 2021-09-01:
In the meantime I'm trying again this way:
void mdnsAndWebHandler()
{ 
#define MDNS_RESTART_TIME 2*60*1000
   static unsigned long lastClientMillis = millis();
   unsigned long ms;
   // 19us  average before first client available
   // 190us average after first client found available
    //### UI on user device browser handler
    mdns.run(); // This runs the mDNS module. CALL THIS PERIODICALLY, OR NOTHING WILL WORK!
    client = server.available();
    if (client) {
        handleClient();
        lastClientMillis = millis();
    }
    ms = millis();
    if ((ms - lastClientMillis) > MDNS_RESTART_TIME)
    {
         mdns.removeAllServiceRecords();
         mdns.addServiceRecord(mDNS_Service.c_str(), 80, MDNSServiceTCP);
         lastClientMillis = ms;  
         dbn("###  Restarted : " + mDNS_Service);
    }
    yield();
}
i.e. every two minutes remove and re-add the service. I'll advise after more tests.

khoih-prog, 2021-09-01:
That's good to use temporarily, if OK, in the meantime before having the correct fix.
This seems to affect too many people, cores, libraries and certainly got lots of attention => correct fix.

khoih-prog, 2021-09-01:
If possible, you can also test with some routers, such as FritzBox, etc., then make a list of `mDNS caching/supporting routers/gateways`. This will be both helpful for many people as well as a push for routers/gateways mfrs to enhance them.

khoih-prog, 2021-09-02:
Hi @stefat 

Please test with the new [**MDNS_Generic** Major releases v1.2.0](https://github.com/khoih-prog/MDNS_Generic/releases/tag/v1.2.0) I believe this issue is fixed.

I tested using ESP32 (ESP-Native WiFi), Nano_RP2040_Connect (WiFiNINA) without any issue.

I can ping using those `hostnames` after many hours without restarting the boards.

Cheers,

---

### Releases v1.2.0

 1. Add support to RP2040-based boards, such as **Nano_RP2040_Connect, RASPBERRY_PI_PICO, ADAFRUIT_FEATHER_RP2040 and GENERIC_RP2040**, using [**Arduino-mbed RP2040** core](https://github.com/arduino/ArduinoCore-mbed) using WiFi / [WiFiNINA_Generic Library](https://github.com/khoih-prog/WiFiNINA_Generic)
 2. Add support to RP2040-based boards, such as **RASPBERRY_PI_PICO, ADAFRUIT_FEATHER_RP2040 and GENERIC_RP2040**, using [**Earle Philhower's arduino-pico** core](https://github.com/earlephilhower/arduino-pico) using WiFi / [WiFiNINA_Generic Library](https://github.com/khoih-prog/WiFiNINA_Generic)
 3. Add support to ESP32 and ESP8266 boards using native ESP WiFi
 4. Add support to Generic boards, such as **MKR WiFi1010**, using [WiFi101 Library](https://github.com/arduino-libraries/WiFi101)
 5. Add WiFi-related examples. 
 6. Modify hostname from easily-duplicated `arduino` to board-related hostname to fix issue caused by duplicated hostname, such as [mDNS stops working after two minutes #5](https://github.com/khoih-prog/MDNS_Generic/issues/5)
 7. Verify issue [**mDNS stops working after two minutes** #5](https://github.com/khoih-prog/MDNS_Generic/issues/5) is fixed, using Nano_RP2040_Connect, WiFiNINA FW 1.4.8 and [WiFiNINA_Generic Library](https://github.com/khoih-prog/WiFiNINA_Generic)
 8. Change License from `MIT` to [GPLv3](https://github.com/khoih-prog/MDNS_Generic/blob/master/LICENSE)

closed 2021-09-02 by khoih-prog

stefat, 2021-09-02:
Hi Khoih,
I tested for 4 hours.
I can connect my client with a very large delay (~30s or more) after few minutes delay with no client access.
I installed your 1.2.0 using Arduino IDE 1.8.15 that asked me to install two dependencies WiFiNINA_generic and another (WyFi101 ? Not sure. I forgot). I accepted to install both. No change in my sketch other than having successfully updated your library (1.2.0 verified).
Any hint ?

BTW please find below my include’s:
#include <WiFiNINA.h>
#include <MDNS_Generic.h>
#include "FlashMem.hpp"
#include "SDCard.hpp"
#include "jsonlib.h"
#include <WDTZero.h> //per il watchDog
#include <arduino-timer.h>
#include <SDU.h> //Secure Digital Update
#include <Scheduler.h>
#include "yasm.h"

Stefano

khoih-prog, 2021-09-02:
You have to start from the [**WiFiRegisteringServices** example](https://github.com/khoih-prog/MDNS_Generic/tree/master/examples/WiFi/WiFiRegisteringServices) to see if it's OK for you first.

Then move ahead by adding your core slowly to know where it's breaking. Something you've done not correctly in your code

Remember to change to use WiFi101 lib for your MKR1010 board

```
#if defined(ESP32) || defined(ESP8266)

  #define WIFI_NETWORK_WIFININA   false
  #define WIFI_NETWORK_WIFI101    false
  #define WIFI_NETWORK_ESPAT      false
  #define WIFI_NETWORK_ESP        true

  String hostname = "ESP_" + String(ESP_getChipId(), HEX);

#else
  // Select one to be true, but WIFI_NETWORK_ESP
  #define WIFI_NETWORK_WIFININA   false
  #define WIFI_NETWORK_WIFI101    true
  #define WIFI_NETWORK_ESPAT      false
  #define WIFI_NETWORK_ESP        false

  String hostname = BOARD_TYPE;

#endif
```

and

```

void loop()
{
  // This actually runs the mDNS module. YOU HAVE TO CALL THIS PERIODICALLY,
  // OR NOTHING WILL WORK! Preferably, call it once per loop().
  mdns.run();

  ...

}
```

khoih-prog, 2021-09-02:
Also post the debug terminal output so that I know if you're doing correctly.

khoih-prog, 2021-09-02:
As I don't have MKR1010 (using WiFi101 lib), if you have any issue with the  [**WiFiRegisteringServices** example](https://github.com/khoih-prog/MDNS_Generic/tree/master/examples/WiFi/WiFiRegisteringServices), use another WiFi board (ESP32, ESP8266, Nano_RP2040_onnect, Nano_33_IoT, etc.) or even boards with Ethernet (SAMD, nRF52, etc.) with [**RegisteringServices** example](https://github.com/khoih-prog/MDNS_Generic/tree/master/examples/Ethernet/RegisteringServices).

It's possible some bug in WiFi101 library, if you have issue with MKR1010. WiFiNINA is tested and OK.

stefat, 2021-09-02:
Ok, done (I hope) as per your suggestion.
1) I used WiFiNINA. WiFi101 at first attempt gave me some trouble.

#if defined(ESP32) || defined(ESP8266)

  #define WIFI_NETWORK_WIFININA   false
  #define WIFI_NETWORK_WIFI101    false
  #define WIFI_NETWORK_ESPAT      false
  #define WIFI_NETWORK_ESP        true

  String hostname = "ESP_" + String(ESP_getChipId(), HEX);

#else
  // Select one to be true, but WIFI_NETWORK_ESP
  #define WIFI_NETWORK_WIFININA   true
  #define WIFI_NETWORK_WIFI101    false
  #define WIFI_NETWORK_ESPAT      false
  #define WIFI_NETWORK_ESP        false

  String hostname = BOARD_TYPE;

#endif

2) WiFiNINA fw: I had 1.4.7 (updated 1 month ago). As your example sketch gave it me stale I updated 1.4.8 (@#!?). It’s ok
3) Run the sketch It gave me tons of
ServerDrv::availData: SpiDrv not available

3)Put in loop() at the top the while(true)

 mdns.run();
 while (true);

Just after the mdns.run I got this log:

18:29:22.539 -> Start WiFiRegisteringServices on SAMD_MKRWIFI1010
18:29:22.539 -> MDNS_Generic v1.2.0
18:29:22.539 -> Used/default SPI pinout:
18:29:22.539 -> MOSI:8
18:29:22.539 -> MISO:10
18:29:22.539 -> SCK:9
18:29:22.539 -> SS:4
18:29:23.322 -> Attempting to connect to SSID: Infostrada-9A8848
18:29:27.765 -> SSID: Infostrada-9A8848
18:29:27.765 -> Local IP address: 192.168.1.138
18:29:27.765 -> Signal strength (RSSI):-73 dBm
18:29:27.765 -> Registering mDNS hostname: SAMD-MKRWIFI1010
18:29:27.765 -> To access, using SAMD-MKRWIFI1010.local
18:29:27.765 -> ServerDrv::availData: SpiDrv not available

COMMENT:
1) MOSI/MISO/SCK look ok for my MKR1010. I’m not sure on SS as in my hw pin 4 and 5 are connected as follows:
const int  PROGMEM SFChipSelect = 5; // digital pin for flash chip CS pin
const int  PROGMEM SDchipSelect = 4; // digital pin for SD card
I never changed the default SPI pin in my sketch.
Maybe this the issue for the example sketch ?
2)WiFiNina 1.4.8 may affect my own sketch with the original problem I submitted to you ? In the meantime I wait for any reply I’m going to test this.

I’m very grateful for your help.

Stefano

khoih-prog, 2021-09-02:
So far so good.

> 18:29:27.765 -> ServerDrv::availData: SpiDrv not available

Ignore this. Just a left-over as I forgot to remove in [WiFiNINA_Generic Library](https://github.com/khoih-prog/WiFiNINA_Generic). Remove that by changing

[server_drv.cpp#L284](https://github.com/khoih-prog/WiFiNINA_Generic/blob/master/src/utility/server_drv.cpp#L284) to

```
#if (KH_WIFININA_SERVER_DRV_DEBUG > 2)
```

It'sOK then, you can issue the `ping` by

```
$ ping SAMD-MKRWIFI1010.local
```

and check if it's continue running after many hours

khoih-prog, 2021-09-02:
I was testing using NINA FW v1.4.8, but I think v1.4.7 is still OK. Must use WiFi101 for MKR1010.

stefat, 2021-09-02:
Great I do that ! I leave it running and be back soon

Stefano

stefat, 2021-09-03:
Test done overnight.

Pattern configuration:
1)The MKRWiFi1010 running WiFiRegisteringServices. I used two devices as clients: a Mac notebook and my iPhone; both with Chrome and Safari browser. In my iPhone I ran the app Net Analyzer that makes net scan and ping.
2) IMPORTANT SIDE NOTE: my Mac has been used (and has been used for long time, also with my original sketch) with its WiFi off and connected to my router/gateway using:
an Ethernet LAN adapter <—>
a power line adapter using my electric home plant wall plugs <—>
router  Ethernet plug.
i.e. NOT using WIFI connection. I think of it as a (slow) Ethernet LAN.
Using the above configuration I usually have:
	-	Mac LAN download connection speed: ~60Mbps;
	-	iPhone Wifi download connection speed: ~20Mbps.
	Measured with the OOKLA app speed test both in Mac and iPhone.
3) Left overnight running quietly. For results and tests schedule see below.

RESULT
A) Test Log:
21:39:25.521 -> Hello from a mDNS-enabled web-server running on your Arduino board!
21:39:27.552 -> Hello from a mDNS-enabled web-server running on your Arduino board!
21:39:29.125 -> Hello from a mDNS-enabled web-server running on your Arduino board!
21:39:30.447 -> Hello from a mDNS-enabled web-server running on your Arduino board!
21:39:33.906 -> Hello from a mDNS-enabled web-server running on your Arduino board!
21:39:43.757 -> Hello from a mDNS-enabled web-server running on your Arduino board!
01:21:47.314 -> Hello from a mDNS-enabled web-server running on your Arduino board!
01:22:03.252 -> Hello from a mDNS-enabled web-server running on your Arduino board!
01:22:03.484 -> Hello from a mDNS-enabled web-server running on your Arduino board!
01:56:06.712 -> Hello from a mDNS-enabled web-server running on your Arduino board!
08:58:00.170 -> Hello from a mDNS-enabled web-server running on your Arduino board!
08:59:14.047 -> Hello from a mDNS-enabled web-server running on your Arduino board!
09:02:24.830 -> Hello from a mDNS-enabled web-server running on your Arduino board!
09:02:57.277 -> Hello from a mDNS-enabled web-server running on your Arduino board!
09:03:52.319 -> Hello from a mDNS-enabled web-server running on your Arduino board!
09:08:26.875 -> Hello from a mDNS-enabled web-server running on your Arduino board!
09:08:27.529 -> Hello from a mDNS-enabled web-server running on your Arduino board!
09:08:29.995 -> Hello from a mDNS-enabled web-server running on your Arduino board!
09:08:47.618 -> Hello from a mDNS-enabled web-server running on your Arduino board!
09:11:26.876 -> Hello from a mDNS-enabled web-server running on your Arduino board!
09:11:26.876 -> Hello from a mDNS-enabled web-server running on your Arduino board!
09:11:30.730 -> Hello from a mDNS-enabled web-server running on your Arduino board!

B) My Mac with Safari, Chrome and pings NEVER failed. Pings from 25ms to 300ms.
C) My iPhone ALWAYS hanged. No answer. Neither pinging. With Safari/Chrome browser and pinging, during the night tests, I quit the tests with no answer. This morning I was astonished (again) watching my iPhone Safari browser hanging the SAMD-MKRWIFI1010.local url with the blue progress bar stopped at top. After not less than ~30s, or more, I was still there and … oops ! The 'hello from a mDNS …’ popped up ! New attempts done just after, did immediately positive  results.

COMMENT
I’m not an expert in wireless communication. I noted this same behaviour testing my original sketch. My Mac with Chrome used for testing my sw rarely (~never, for a month or so) failed connecting. My iPhone and iPad always gave troubles ! Sometimes, rarely, they connected.
Maybe the udp fails e.g. with a poor WiFi connection ? And sometimes works after few attempts ? And it anyway may ask for epic times due to e.g. retries ? All this caused by the methods the various hw/sw's involved implement to make the mdsn broadcast requests and answers.

Your comment highly appreciated.

Stefano

stefat, 2021-09-03:
BTW: what is the difference between the
#include <WiFiNINA.h> (I’ve been using in my sketch. No library for WiFiUdp. It's included in the WiFiNINA)
and
#include <WiFiNINA_Generic.h>
#include <WiFiUdp_Generic.h>
?

I’d say this is the single difference I see between my sketch and the test one.

Stefano

khoih-prog, 2021-09-03:
Hi.

> 
> RESULT
> A) Test Log:
> 21:39:25.521 -> Hello from a mDNS-enabled web-server running on your Arduino board!
> 21:39:27.552 -> Hello from a mDNS-enabled web-server running on your Arduino board!
> ...
> 
> 09:11:26.876 -> Hello from a mDNS-enabled web-server running on your Arduino board!
> 09:11:30.730 -> Hello from a mDNS-enabled web-server running on your Arduino board!

> B) My Mac with Safari, Chrome and pings NEVER failed. Pings from 25ms to 300ms.

So this is OK now for you and the issue is fixed.

> C) My iPhone ALWAYS hanged. No answer. Neither pinging. With Safari/Chrome browser and pinging, during the night tests, I quit the tests with no answer. This morning I was astonished (again) [...]

It's also OK, but somehow iPhone + Safari don't behave as expected like the other platforms.
Nothing we can do here. You'd better ask Apple and/or Apple Forum for help.

> #include <WiFiNINA.h> (I’ve been using in my sketch. No library for WiFiUdp. It's included in the WiFiNINA)
> and
> #include <WiFiNINA_Generic.h>
> #include <WiFiUdp_Generic.h>

---

You have to use my [WiFiNINA_Generic library](https://github.com/khoih-prog/WiFiNINA_Generic), which is a modified version of [WiFiNINA](https://www.arduino.cc/en/Reference/WiFiNINA) as follows

```
 #include <WiFiNINA_Generic.h>
 #include <WiFiUdp_Generic.h>
```

This way, the correct WiFi101 library will be used for your MKR WiFi1010

stefat, 2021-09-03:
@khoih thank you very much for everything. Highly appreciated. Keep up your good work.