khoih-prog/MDNS_Generic #11: Does not advertise properly on stm32

gigaj0ule, 2022-06-20:
I have been trying out the library on STM32F103 with the wiznet 5500 chip, and Ethernet_Generic library.

I'm able to discover services well, and my QM packets are answered with the responses processed nicely.

```
27755	509.550094542	10.0.0.242	224.0.0.251	MDNS	86	Standard query 0x0000 PTR _http._tcp.local, "QM" question
```

```
Discovering services of type '_http' via Bonjour...
Found: 'EPSON XP-810 Series' at 10.0.0.225:80
Found: '_printer' at 10.0.0.110:23595
Found: '_ipp' at 10.0.0.110:2049
Found: '_http' at 0.0.0.0:16723
Found: 'EPSON XP-810 Series' at 10.0.0.225:80
Found: '_scanner' at 10.0.0.225:2049
Found: '_ipp' at 0.0.0.0:16723
Found: 'EPSON XP-810 Series' at 10.0.0.225:80
Found: 'EPSON XP-810 Series' at 10.0.0.225:80
Found: '_ipp' at 0.0.0.0:16723
Finished discovering services of type _http
```

Curiously the service browser returns more than _http...

But more seriously, I cannot seem to get my service recognized by mdns-scan

I have confirmed that wireshark can see my packets:

```
27996	512.433768171	10.0.0.242	224.0.0.251	MDNS	291	Standard query response 0x0000 SRV, cache flush 0 0 80 arduino.local TXT, cache flush PTR _http._tcp.local PTR arduino._http._tcp.local A, cache flush 10.0.0.242```

But perhaps they are malformed?

gigaj0ule, 2022-06-20:
More clues: in avahi-browse i'm seeing what looks like buffer overrun in the name 0_o

```
+ wlp3s0 IPv4 my_arduino                                    _http\028._tcp       local
```

khoih-prog, 2022-06-20:
Hi @gigaj0ule 

> I'm able to see arduino with avahi-browse, but not mdns-scan

Can you check to be sure `mdns-scan` is fully compatible and OK to use? Also check using other better boards (STM32F4/F7, etc), or the original [**ArduinoMDNS**](https://github.com/arduino-libraries/ArduinoMDNS) library, if possible, to isolate the issue (STM32 F1 board, Ethernet, this library's bug, etc.)

Anyway, I'll archive this library soon as I won't have time to spend on this library anymore.

Your investigation, updated info, PR are appreciated to help other users.

Best Regards,

gigaj0ule, 2022-06-20:
I need to do more looking into mdns-scan to know where the bug is exactly. 

I'll keep digging! 

As this is the only ethernet-compatible mDNS library for Arduino, I am not sure why you would want to archive it?

khoih-prog, 2022-06-20:
> As this is the only recently-maintained ethernet-compatible mDNS library for Arduino, I am not sure why you would want to archive it?

I'm sorry I really don't have time to spend on this library and mDNS topic for quite a long time. It's hard and time-consuming to start over every time to reread all the necessary info / specs.

Hopefully someone (you ???) will take over and maintain this library. Before that time, I'm still trying my best and really appreciate any contribution (bug fix, PR, etc.)

It's good you're making some progress.

gigaj0ule, 2022-06-20:
Time ah yes, the one thing I can't buy more of x_x

Well good news is... it looks like this unmerged PR fixes the overrun problem I just noticed!

https://github.com/arduino-libraries/ArduinoMDNS/pull/8/commits/396aef16a2826e6f3207f7444251eae322251ca5

\028 is no longer reported at the end of _http

khoih-prog, 2022-06-20:
If tested OK and problem fixed, could you make the PR? Thanks

gigaj0ule, 2022-06-20:
Not sure entirely, I still have more acid tests to run before I'm satisfied that these bugs are fixed :-)

closed 2022-10-11 by khoih-prog