khoih-prog/MDNS_Generic #13: UDP Length check discards correct responses when resolving names - On Nano RP2040 Connect

2001herne, 2022-10-12:
Arduino IDE version: 2.0.0
MbedOS core v3.3.0
Attempting to resolve host names belonging to the following:
* Raspbian linux running avahi 0.8
    *  output of `uname -a` is:  `Linux piNetHub 5.10.63-v7l+ #1488 SMP Thu Nov 18 16:15:28 GMT 2021 armv7l GNU/Linux`
* Arch Linux running avahi 0.8
    * output of `uname -a` is `Linux archGaming 5.19.13-arch1-1 #1 SMP PREEMPT_DYNAMIC Tue, 04 Oct 2022 14:36:58 +0000 x86_64 GNU/Linux`
Network config: All devices are connected to a WPA2 Personal network operating on 2.4 GHz
### Describe the bug

When resolving a hostname, I receive a response with a length of 44. Disabling the length check results in me obtaining the correct IP address. This occurs on a Arduino Nano rp2040 connect, running Arduino MbedOS Nano. I am able to resolve this hostname using `avahi-resolve-host-name` on my Linux laptop.

### Steps to Reproduce

1. Upload the `WiFiResolvingHostnames.ino` sketch
2. Attempt to resolve hostname
3. Output is `Resolving '...' error`
4. Modifying `MDNS_Generic_Impl.h` to remove the check for UDP length correctly resolves the hostname.

### Expected behavior

For a valid host name to resolve to the associated IP address.

### Actual behavior

The response (containing a correct IP address) is discarded by the 48 length check.

khoih-prog, 2022-10-12:
Hi @2001herne 

Thanks for spotting the bug. I'll fix and have a new release soon.

Best Regards,

khoih-prog, 2022-10-13:
Hi @2001herne  

The new [**MDNS_Generic** v1.4.2](https://github.com/khoih-prog/MDNS_Generic/releases/tag/v1.4.2) has just been published. Your contribution is noted in [**Contributions and Thanks**](https://github.com/khoih-prog/MDNS_Generic#contributions-and-thanks)

Best Regards,

---

### Releases v1.4.2

1. Fix bug in UDP length check and in WiFi example. Check [UDP Length check discards correct responses when resolving names - On Nano RP2040 Connect #13](https://github.com/khoih-prog/MDNS_Generic/issues/13)
2. Fix bug in example [WiFiDiscoveringServices](https://github.com/khoih-prog/MDNS_Generic/tree/master/examples/WiFi/WiFiDiscoveringServices)
3. Update `Packages' Patches`

closed 2022-10-13 by khoih-prog